tenstorrent/tt-umd #26: Move common device/driver implementations to base `tt_device` class

vtangTT, 2024-08-05:
```
std::uint64_t tt_SimulationDevice::get_pcie_base_addr_from_device()
virtual void set_device_l1_address_params(const tt_device_l1_address_params& l1_address_params_);
virtual void set_device_dram_address_params(const tt_device_dram_address_params& dram_address_params_);
virtual void set_driver_host_address_params(const tt_driver_host_address_params& host_address_params_);
virtual void set_driver_eth_interface_params(const tt_driver_eth_interface_params& eth_interface_params_);
```

These ones are common among `tt_SiliconDevice`, `tt_SimulationDevice`, `tt_EmulationDevice`. There might be more.

FYI: @joelsmithTT

joelsmithTT, 2024-08-05:
There may be fields in the `tt_..._params` structures that are unused, so there's an opportunity for some cleanup/consolidation here.

Also related: the error messages from `tt_SiliconDevice` are bad if a code path expects `set_driver_eth_interface_params` to have been previously called.

I think there is an opportunity here to make `tt_SiliconDevice` more difficult to use incorrectly.

broskoTT, 2024-09-19:
Let's wait a bit to get clarity from #57 
The future design does address having a base Chip class with common implementation (In the design, tt_SiliconDevice = LocalChip + RemoteChip, tt_SimulationDevice = VersimChip)

broskoTT, 2024-11-07:
This will be naturally solved by creating Chip class, moving simulation and mockup to be a specialization of Chip, and removing tt_device.
Will keep this open just in case.

broskoTT, 2025-03-03:
Naturally solved by adjusting set_ functions.
Further when tt_device is completely removed it will also be further solved for set_barrier_params as well.

closed 2025-03-03 by broskoTT