tenstorrent/tt-umd #32: Consolidate and cleanup tests folder

vtangTT, 2024-08-16:
Right now, UMD unit tests for different chips are found in different folders (eg: `tests/wormhole`, `tests/blackhole`)

Each folder has the same file(s): `test_silicon_driver.cpp`, `test_umd_remote_api_stability.cpp`

These files seem to be just copy-pasted and share a lot of common code. Should be able to consolidate all of them with some sort of gtest fixture, get rid of the separate folders and cleanup the cmake build setup.

This might change completely when UMD gets refactored, but just tracking with this ticket.

vtangTT, 2024-08-16:
This will probably break the gitlab CI.

pjanevskiTT, 2024-11-05:
Putting this as P2 as it might completely change when we redesign the API and have to write new set of tests

closed 2025-03-10 by broskoTT