tenstorrent/tt-umd #53: Consistent logging, with a proper logger module

broskoTT, 2024-09-10:
Per chat with Joel:

Consistent logging, with a proper logger module (there are a few different logging techniques in UMD, including a home-grown logger that looks like it came from BBE).  Something that gives us timestamps, and source location (file.cpp:123) would be nice.  I've used spdlog (fmt-based) and glog (stream style syntax) in the past but I don't have strong opinions about the library.  Something that's thread safe and fast, can log to stderr or to a file.

vtangTT, 2024-09-17:
Related to this, we should get rid of `LOG1` `LOG2` and similar stuff

blozano-tt, 2024-10-09:
I vote for `spdlog`

broskoTT, 2024-10-09:
> I vote for `spdlog`

Is this used by tt-metal as well? I'd like us to be consistent.

closed 2024-10-10 by pjanevskiTT