tenstorrent/tt-umd #55: Consistent error handling

broskoTT, 2024-09-10:
Per chat with Joel:

Consistent error handling.  UMD is using exceptions but I doubt anybody is catching them.  We also have a log_assert.  A pattern I've seen used successfully is to make macros for throwing an exception or terminating the program if it its a fatal error, so that the exact source location can be included in the exception what() or written to stderr.  UMD_THROW("That won't work") or UMD_FATAL("Grayskull is on fire").  On exceptions, if we're going to use them then I think we should be explicit in interface documentation what methods can throw and why.

vtangTT, 2024-09-20:
I feel like we should use the same error handling as metal does rn with `TT_FATAL`, `TT_ASSERT`, etc. 

There might be an opportunity to consolidate these utility functions as metal and umd both define their own rn in `common/assert.hpp` or `common/logger.hpp` and it gets confusing to tell which one people are including.

Maybe we can consider a third party utils repo that gets included in both metal and umd as a submodule so we don't have duplicate code?

broskoTT, 2024-09-23:
@vtangTT that makes sense. But the current issue is scoped only to make uniform error handling within this repo, I'd like to keep it like that for know.

I'd advise to create another issue for what you mentioned, although I'm not sure whether that should be tt-umd issue or tt-metal issue or something else. More of the problem is to actually create a new repo, then to move tt-metal to use it. I think it will be a minor task to make umd point to it as well.

pjanevskiTT, 2024-11-05:
I will close #139 since this issue has more details, @vtangTT can you take this over when you have time?

broskoTT, 2025-07-21:
Scope of this issue, figure out what should we use, which macro, should we stop using log_error or log_assert, etc.
Then make everything uniform in our repo.

aleksamarkovicTT, 2025-12-25:
Yes I agree. The only thing I'm not sure to use TT_THROW for is [out_of_range](https://github.com/tenstorrent/tt-umd/blob/bc14a3b5f3ae3b13092cfcdc1a596ee250d316ce/device/pcie/tlb_window.cpp#L139). What do you think?

pjanevskiTT, 2025-12-25:
Completely fine with it