tenstorrent/tt-umd #56: Separate arch specific code to directories

pjanevskiTT, 2024-09-10:
Separate arch specific code into separate directories. This way we achieve two things
- Avoid conditional compilation (#51) -> we can have build target for each arch
- Code organization looks much better

For example, for something like https://github.com/tenstorrent/tt-umd/blob/main/device/driver_atomics.h, we should have something like

```
arch
|----- x86
|       |------ driver_atomics.cpp
|
|----- arm
|       |------ driver_atomics.cpp
```

Maybe driver atomics file is not the right granularity to do this, maybe we need to do it in smaller/bigger chunks, but that should be investigated through this issue as well.

pjanevskiTT, 2024-11-05:
We already had one PR closed on this because it adds complexity, I won't close this because it would be good to have this sometime in the future, but it is really low prirority at the moment

broskoTT, 2025-07-21:
When this pops up again as an issue, we'l just open the issue at that point. 
The scope of this issue currently is undefined, referencing potential future needs.