tenstorrent/tt-umd #73: Revisit create-ethernet-map

broskoTT, 2024-09-24:
Investigate whether it is possible to have the code present in the repo, and build it with umd, rather than having a binary. Motivation is that this would make it easier to flag all the binaries we need to ship to the customer.

pjanevskiTT, 2024-11-05:
Improved with #245, next step would be to have source code directly inside tt-umd repo

broskoTT, 2025-03-10:
Already working on this as part of: #560 #481 #459 
However, we currently don't have a plan to move old wormhole code to CEM, but the situation today is much better.

pjanevskiTT, 2025-05-08:
Main work of having CEM from source is done by #722, follow up issues to guarantee safe introduction to main code path are open, so closing this one

closed 2025-05-08 by pjanevskiTT