tenstorrent/tt-umd #77: reading/writing more than 1KB of data from second chip on N300 fails

tt-vjovanovic, 2024-09-25:
We are using UMD in tt-debuda and having trouble when reading/writing more than 1KB of data from second chp on N300. Either our initialization of UMD is bad or there is a bug in UMD.
In function `tt_SiliconDevice::read_from_non_mmio_device` there is condition `use_dram = size_in_bytes > 1024;`...
We are using `tt_SiliconDevice::read_from_device` and `tt_SiliconDevice::write_to_device`...

joelsmithTT, 2024-09-25:
@tt-vjovanovic can you share a git commit hash of your repo that demonstrates the problem?

broskoTT, 2024-09-27:
Metal has this defined
const uint32_t MAX_BLOCK_SIZE = (0x1 << 10); // Max 1024 bytes

tt-vjovanovic, 2024-09-30:
> @tt-vjovanovic can you share a git commit hash of your repo that demonstrates the problem?

3bff9d422e35b8ff2e28f99d6ec63eb31d00cf1e third_party/umd

tt-vjovanovic, 2024-09-30:
I'm not saying this commit caused the problem, but that is where our submodule points to :)

broskoTT, 2024-10-21:
Related, to be fixed after this issue is resolved: https://github.com/tenstorrent/tt-debuda/issues/124