tenstorrent/tt-umd #82: Fix n300 unit test failing

broskoTT, 2024-09-26:
Could be due to the specific machine. Verify if the same test passes on tt-metal first.
This test is failing on each new PR at the moment.

pjanevskiTT, 2024-09-26:
Just to add more information, the test failing is `SiliconDriverWH.HarvestingRuntime` from [test_silicon_driver_wh.cpp](https://github.com/tenstorrent/tt-umd/blob/main/tests/wormhole/test_silicon_driver_wh.cpp#L134)

joelsmithTT, 2024-10-04:
```
[ RUN      ] SiliconDriverWH.HarvestingRuntime
2024-10-04 18:43:35.733 | INFO     | SiliconDriver   - Detected 1 PCI device : [0]
2024-10-04 18:43:35.735 | WARNING  | SiliconDriver   - bind_area_memory_nodeset(): Unable to determine TT Device to NumaNode mapping for physical_device_id: 0. Skipping membind.
---- ttSiliconDevice::init_hugepage: bind_area_to_memory_nodeset() failed (physical_device_id: 0 ch: 0). Hugepage allocation is not on NumaNode matching TT Device. Side-Effect is decreased Device->Host perf (Issue #893).
2024-10-04 18:43:35.750 | INFO     | SiliconDriver   - Software version 6.0.0, Ethernet FW version 6.9.0 (Device 0)
2024-10-04 18:43:35.751 | INFO     | SiliconDriver   - Software version 6.0.0, Ethernet FW version 6.9.0 (Device 1)
../tests/wormhole/test_silicon_driver_wh.cpp:195: Failure
Expected equality of these values:
  vector_to_write
    Which is: { 0, 1, 2, 3, 4, 5, 6, 7, 8, 9 }
  dynamic_readback_vec
    Which is: { 0, 1, 2, 3, 4, 5, 6, 7, 8, 0 }
Vector read back from core 1-7does not match what was written
[  FAILED  ] SiliconDriverWH.HarvestingRuntime (85 ms)
```
from https://github.com/tenstorrent/tt-umd/actions/runs/11185110395/job/31097527048?pr=114

joelsmithTT, 2024-10-07:
It passes at least some of the time.

Fail: https://github.com/tenstorrent/tt-umd/actions/runs/11185110395/job/31190146087?pr=114
Pass: https://github.com/tenstorrent/tt-umd/actions/runs/11185110395/job/31191898851?pr=114

appears to be the same machine `umd-n300-1`.  I'm trying to replicate.

joelsmithTT, 2024-10-08:
Test in question has been disabled to unblock CI: https://github.com/tenstorrent/tt-umd/pull/132

@broskoTT, do you think it's worth swapping out `umd-n300-1` for a different machine for CI?  I'd like to try reproducing the problem on the machine that's affected.

broskoTT, 2024-10-09:
> I'd like to try reproducing the problem on the machine that's affected.

You need access to cloud VPN I think to access cloud machines directly. But I'd expect you'd see the same thing as we're seeing from the CI runs. I think it could be the easiest to swap out the machine, but I'm still wondering why is this machine failing, wrong harvest info or something else? Should we flag it so others don't use it as well?

Note that the CI was not blocked by failure of this test. I intentionally didn't want to set these machine tests as mandatory so that PRs are not blocked if our machines fail for any reason, or jobs are blocked, etc.

joelsmithTT, 2024-10-09:
> Note that the CI was not blocked by failure of this test

I might be misremembering but I think I needed a successful run of all the tests before I got https://github.com/tenstorrent/tt-umd/pull/114 merged.

joelsmithTT, 2024-10-09:
>  I'm still wondering why is this machine failing, wrong harvest info or something else? Should we flag it so others don't use it as well?

I want to do some hands-on debugging with this particular machine/device to understand what is wrong.

broskoTT, 2025-04-03:
#683 might be another instance of this. Recheck if this is still the problem after that is resolved.