tenstorrent/tt-umd #91: Harvesting

joelsmithTT, 2024-09-30:
From @abhullar-tt: "Throughout Metal and UMD we assume only rows are harvested. On BH columns will be harvested."

abhullar-tt, 2024-10-01:
For initial support of column harvesting it seems like we need to:
- use create-eth-map utility on BH to discover harvesting masks
- uplift to interpret this as columns harvested on BH rather than rows 

@broskoTT is this a good time to uplift the soc desc to have all the coordinate translations?

pjanevskiTT, 2024-11-05:
This has been partially solved with #202 , we just need integration of the changes inside tt_SiliconDevice

broskoTT, 2024-11-06:
The only thing left now is #115 , right? We can close this issue?

pjanevskiTT, 2024-11-06:
We can, I have smaller issues to address this

closed 2024-11-06 by pjanevskiTT