tenstorrent/tt-umd #95: Non-MMIO targets for Blackhole

pjanevskiTT, 2024-10-01:
Non-MMIO targets (multichip) are not supported for Blackhole at the moment.

Not really sure what are the limitations at the moment, if there are any. When we were bringing up new features for Blackhole in UMD, we only had single chip cards, so there was no need to enable multichip.

Not sure whether this is just an implementation thing, I think there shouldn't be any fundamental limitation - (ETH firmware possibly ?)

abhullar-tt, 2024-10-01:
I believe that eth firmware is not ready yet.

broskoTT, 2024-11-06:
@abhullar-tt has this changed?

pjanevskiTT, 2025-01-13:
@broskoTT do we even have a Blackhole product which is going to require non-mmio target from the host? On p300 both chips have PCIE, galaxy as well

abhullar-tt, 2025-01-13:
> [@broskoTT](https://github.com/broskoTT) do we even have a Blackhole product which is going to require non-mmio target from the host? On p300 both chips have PCIE, galaxy as well

No, on BH all chips will have pcie connections. I think this issue can be closed. BH would never hit code paths where host does non-MMIO device writes/reads

pjanevskiTT, 2025-01-13:
That is what I thought, thanks Almeet

closed 2025-01-13 by pjanevskiTT