tenstorrent/tt-umd #100: SoCDescriptor

broskoTT, 2024-10-03:
Related to https://docs.google.com/drawings/d/1-m1azdsBqMA0A6ATYRMfkhyeuOJuGCEI62N5a96LXj0/edit?usp=sharing
This should not be something which is given to UMD, rather provided by UMD.
Remove passing socDescriptor to constructor of tt_SiliconDriver.
Add default SoCDescriptors (from tt-metal repo) to UMD repo. Construct one of these based on arch type reported by TTDevice/PCIDevice class.

See if some methods from metal_SocDescriptor make sense to be moved here.

broskoTT, 2024-10-09:
Per chat on the main channel, make sure to leave the variant with passing a custom soc descriptor in case it is needed for some custom testing

broskoTT, 2024-10-09:
@pjanevskiTT per chat from today we've had. If you're enthusiastic, you can try to embed default soc_descriptor files while building, using "file" and "configure_file" commands from cmake to embed the contents of soc_desciptors directly into c++ code. Then the actual binary won't need files in the right place alongside it.

pjanevskiTT, 2024-11-18:
https://github.com/tenstorrent/tt-umd/issues/314 separate issue for paths to soc descriptors, this one was mostly about removal of soc descriptor from Cluster constructor, so closing it since it was resolved by #277

closed 2024-11-18 by pjanevskiTT