tenstorrent/tt-umd #117: Improve device enumeration code

joelsmithTT, 2024-10-04:
tt_silicon_driver.cpp has `std::vector<chip_id_t> ttkmd_scan()` (exposed as `static std::vector<chip_id_t> tt_SiliconDevice::detect_available_device_ids();`).

cpuset_lib.cpp has its own PCIe device enumeration code, which has caused me problems in the past in mutli-card systems with one or more cards unbound from the tenstorrent kernel driver.

Finally, `create-ethernet-map` binary does device enumeration both locally (PCIe) and across the topology of Ethernet-connected chips.

Some considerations:
* Ideal: a single interface for PCIe device enumeration, single interface for topology discovery.
* Get rid of `create-ethernet-map`: https://github.com/tenstorrent/tt-umd/issues/73
* Long term, can we skip the topology information going through a YAML phase and instantiate the right Chip instances directly?
* Get rid of ARCH_NAME when building unit tests.

broskoTT, 2024-10-09:
> * Get rid of ARCH_NAME when building unit tests.

This will be covered through some other issue.

> * Ideal: a single interface for PCIe device enumeration, single interface for topology discovery.
I do agree device_id discovery should be in single place.
As is also described (briefly) on https://docs.google.com/drawings/d/1-m1azdsBqMA0A6ATYRMfkhyeuOJuGCEI62N5a96LXj0/edit
The idea is for this to be behind static functions tt::umd::ClusterDescriptor::GetClusterDescriptor() and tt::umd::ClusterDescriptor::GetAllChips() and cluster_descriptor->get_mmio_chips()

pjanevskiTT, 2024-11-05:
Closing this just because it is already broken up into multiple smaller issues

closed 2024-11-05 by pjanevskiTT