tenstorrent/tt-umd #137: Remove notion of logical id from PCIDevice class

joelsmithTT, 2024-10-09:
It doesn't belong there.

broskoTT, 2024-10-10:
I've left a comment in the PR as well, but want to make this clear here as well:
Is device ID guaranteed to be the same as logical ID ?

Having a well thought and consistent naming should be enough.

soc_id -> since we detect soc_descriptor based on it anyway
dev_id -> it is in /dev folder
chip_id -> it corresponds to one chip

I don't even think that these are the best names, just suggested some naming scheme.

joelsmithTT, 2024-10-10:
> Is device ID guaranteed to be the same as logical ID ?

Not guaranteed - I can get my system into a state where they are not.

Most of the time it is probably a reasonable assumption that with N PCIe chips and M total chips, 0 <= dev_id < N and 0 <= chip_id < M, with the bottom N elements of both equal and referring to the same chip.

But this behavior is not guaranteed by KMD and `create-ethernet-map`.  There is a more systemic issue here in that UMD relies on an undocumented tool to produce undocumented output which UMD parses to know what chip is where.

pjanevskiTT, 2024-11-05:
@broskoTT @joelsmithTT what is the state of this?

joelsmithTT, 2024-11-05:
Still needs work.  There are APIs where it isn't clear what the right type is.

broskoTT, 2024-11-06:
This was somewhat fixed in #149 . Looks like it might still need work.

broskoTT, 2025-03-07:
This has been somewhat fine for some time, maybe another pass through code to make sure that all ids have any kind of prefix, which points to the type of id.

pjanevskiTT, 2025-03-07:
I think this is going to become interesting if we need to expose unique chip IDs to, for example, tt-distributed. Not sure if they would like to combine two Clusters (probably yes), so for internal usage we need to find some kind of unique chip ID. I think for Blackhole this can be sorted as board_id + asic location, I know for Wormhole this is a bit messy at the moment. We can discuss this in a separate issue maybe as well.

broskoTT, 2025-07-21:
I don't think there is anything left to do on this particular issue.