tenstorrent/tt-umd #140: Current logging situation allows broken code

joelsmithTT, 2024-10-09:
logger.hpp says,

```
#else

#define log_trace(...) ((void)0)
#define log_debug(...) ((void)0)
#define log_profile(...) ((void)0)
```

This allows for broken log messages:

```
log_debug("Message: {}", v); // v doesn't exist
```

The above shouldn't ever compile.

broskoTT, 2024-11-06:
@pjanevskiTT did we end up figuring out how to turn on logging in UMD?

broskoTT, 2025-04-03:
At one point we should take the time to figure out the logging part of UMD, possibly related to #517

joelsmithTT, 2025-04-06:
I fixed this particular problem in https://github.com/tenstorrent/tt-umd/pull/604

joelsmithTT, 2025-04-06:
Oops - I fixed the instances of this problem in #604.  The possibility for introducing such defects remains.

pjanevskiTT, 2025-12-25:
@broskoTT what is the state of this?