tenstorrent/tt-umd #142: Lower PCIDevice-specific operations to PCIDevice class

joelsmithTT, 2024-10-09:
The `tt_SiliconDevice` class contains data structures that are indexed by the so-called "logical" id, yet some of these structures represent things that conceptually do not exist for remote chips.

From this standpoint, these structures would make more sense if they were indexed by "device id" (not to be confused with PCIe device id).  However, a better design would move these concepts out of the `tt_SiliconDevice` class and into `PCIDevice`.

broskoTT, 2024-11-04:
I'm going through some rewrite... here are the items that make sense to be moved:

- bar io: bar_write32 and bar_read32
- iatu specific code: iatu_configure_peer_region and init_pcie_iatus and code from destructor
- hugepages: lots of code, init_hugepage, also destruction. Also moves out the communication with the kernel driver.
- arc_msg: There is also remote_arc_msg. Some part should live in some lower part of the stack, and some part has to be part of the Chip abstraction.

pjanevskiTT, 2024-11-05:
@broskoTT situation here?

broskoTT, 2024-11-06:
> [@broskoTT](https://github.com/broskoTT) situation here?

In progress.
#253 resolved hugepages part.
The other 3 items still remain.