tenstorrent/tt-umd #187: tt_simulation_host does not abstract away implementation details

blozano-tt, 2024-10-18:
We are currently exposing our implementation details to consumers.

nng is a public dependency, because we don't separate interface and implementation.

Can we make this more opaque? @vtangTT 

@broskoTT are you already planning for this?

https://github.com/tenstorrent/tt-umd/blob/f7b1ce0f6ed3101bdd4f1367145378a59996f07b/device/simulation/tt_simulation_host.hpp#L7-L8

vtangTT, 2024-10-18:
Will look into this next week.

broskoTT, 2024-10-22:
I didn't think about it until recently (the whole simulation class). Will bear this in mind.
@blozano-tt  Did you come to a consensus with the metal folks on layout? Should it be include/public_headers.h and src/private_headers. and src/private_impl.cpp and src/public_impl.cpp?

vtangTT, 2024-10-31:
https://github.com/tenstorrent/tt-umd/pull/224 closes this issue.

closed 2024-10-31 by vtangTT