tenstorrent/tt-umd #197: Use cmake install in CI

blozano-tt, 2024-10-22:
can we switch now to cmake install in CI which builds tests? I couldn't make this work, and then I ended up manually listing all the files required to a tarfile. 
Not saying this has to be part of this PR, but create an issue if it is left to be done.

_Originally posted by @broskoTT in https://github.com/tenstorrent/tt-umd/pull/192#discussion_r1810187753_

pjanevskiTT, 2024-11-05:
@blozano-tt status?

blozano-tt, 2024-11-05:
> [@blozano-tt](https://github.com/blozano-tt) status?

We have not defined test cases to be installed.

CI executes tests directly from the build tree (not the install tree).

I have not prioritized this.

pjanevskiTT, 2024-11-06:
No worries, just wanted to know that I don't have to close the issue, keeping it as a P2 feature (low priority)

broskoTT, 2025-09-18:
We haven't tested cmake install properly.
We've been hitting problems with missing headers a couple of times, specifically in metal build, but now also Vuk is hitting this now with some prototyping.

The task for this work is to restructure CI so that we build some targets (like examples, or tools, or nanobind) strictly against install folder, which will then catch improper public header configuration (among other problems).

Specific issue that Vuk reported: 

```
In file included from /localdev/vjovanovic/umd_install/include/umd/device/tt_device/tt_device.hpp:14,
                 from /localdev/vjovanovic/umd_install/include/umd/device/chip/chip.hpp:12,
                 from /localdev/vjovanovic/umd_install/include/umd/device/cluster.hpp:21,
                 from test_umd.cpp:7:
/localdev/vjovanovic/umd_install/include/umd/device/arc/arc_telemetry_reader.hpp:13:10: fatal error: umd/device/types/telemetry.hpp: No such file or directory
   13 | #include "umd/device/types/telemetry.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

Not sure why we don't already hit problems, during wheel build for example.