tenstorrent/tt-umd #200: Add pre-commit to CI

blozano-tt, 2024-10-22:
Probably easiest with this: https://github.com/pre-commit/action
Then the GitHub action will do the same thing as what runs locally, but catches pushes from a client without pre-commit configured and those that pushed with `--no-verify`.  It'll run slower since it'll check all files, I think.  Maybe some extra configuration to compare against `main` and find a subset to run against.  But if it's fast enough that extra legwork may not be warranted.

_Originally posted by @afuller-TT in https://github.com/tenstorrent/tt-umd/pull/192#discussion_r1810964098_

afuller-TT, 2024-11-05:
If someone wants to do it on UMD, should be able to just copy/paste these lines from Metal: https://github.com/tenstorrent/tt-metal/blob/main/.github/workflows/all-static-checks.yaml#L11-L35
I've had that running for a little while now and seems good.

blozano-tt, 2024-11-05:
Closed as completed in #203

closed 2024-11-05 by blozano-tt