tenstorrent/tt-umd #226: Cluster doesn't support multiple unconnected clusters

broskoTT, 2024-10-29:
A follow up from #165 
Due to the way get_closest_mmio_capable_chip works, and due to the way create-ethernet-map works, it doesn't work correctly for multiple unconnected clusters.
Copying some details from the chat:

It only uses eth_coord_t data for each chip to determine this. It looks to me like this code assumes that all the chips are connected in some way, and doesn't handle two separate clusters of chips (like three N300 cards that I have). So the code currently returns the same chip for any remote chip (so it will return an unconnected local chip sometimes).

Daniel Rosen: But imo we don’t need to worry too much about these corner cases, I think that tt-fabric is looking to be ready in December-ish (if not sooner) and that’ll completely change the cem and coordinate requirements

pjanevskiTT, 2024-11-05:
@broskoTT label and assign priority for this please, not sure about it

broskoTT, 2024-11-06:
@abhullar-tt raised that they need this for clusters with multiple blackholes, bumping priority

closed 2024-11-18 by broskoTT