tenstorrent/tt-umd #229: Cannot detect multiple devices on 2x p150 machine

abhullar-tt, 2024-10-30:
yyzo-bh-01 is 2x p150 but running `tt_SiliconDriver::detect_available_device_ids()` only returns one device. Not sure if this is a UMD issue or something wrong with the machine.

broskoTT, 2024-10-30:
I'm detecting 2 devices.
`ls /dev/tenstorrent` also returns 2 devices
This was mu reservation command for ird, did you indeed take 2 cards?
`ird reserve --docker-image ghcr.io/tenstorrent/tt-metal/tt-metalium/ubuntu-20.04-amd64:v0.52.0-rc4-dev --timeout max blackhole --machine yyzo-bh-01 --num-pci 2`

Note that create-ethernet-map still doesn't work for blackhole, so somewhere in the code we might just assume 1 card in the cluster. But enumerate_device, which just lists /dev/tenstorrent seems to be working.

Can you double check, and if it still doesn't work please provide steps for reproducing the issue

abhullar-tt, 2024-10-30:
Ah user error, I didn't add `--num-pci 2` last night when I tried this. But I confirmed that create-eth-map does not work for BH

closed 2024-10-30 by abhullar-tt

broskoTT, 2024-10-31:
Will be supported after #235