tenstorrent/tt-umd #241: create-eth-map for Blackhole not working

pjanevskiTT, 2024-11-01:
Introduced by #235, create-eth-map binary is not working fully as expected. Chips section of the cluster descriptor is empty, it doesn't show any chips. The binary still works for Wormhole (Grayskull not tested)

I am not really sure how this is built from source, but I know tt-debuda has a binary in their own repo, which I have tested and it produced valid results. This binary is for Blackhole only, so we couldn't reuse it as a general binary, but maybe we could replicate their way of using these binaries

For reference, I am attaching three files (txt files to be able to upload it here)

1. cluster descriptor that binary from tt-debuda produces - 
[cluster_debuda.txt](https://github.com/user-attachments/files/17598278/cluster_debuda.txt)
2. cluster descriptor that we have used as a hardcoded for blackhole so far - https://github.com/tenstorrent/tt-umd/blob/main/blackhole_1chip_cluster.yaml
3. cluster descriptor that updated binary in tt-umd produces - 
[cluster_umd.txt](https://github.com/user-attachments/files/17598279/cluster_umd.txt)

TTDRosen, 2024-11-01:
Hey! That's operating as expected i.e. consistent with GS. Between the debuda version and the release the decision was made to remove coordiantes from Blackhole so there was nothing for me to put into the chips field. This is consistent with Grayskull which also doesn't have coordinates.

broskoTT, 2024-11-05:
I've reopened issue. But not with the task to fix CEM, rather to fix yaml parsing in tt_ClusterDescriptor.
This might unlock also not having specific grayskull codepath.

abhullar-tt, 2024-11-05:
I  started some changes in `abhullar/bh-cluster-desc` to parse a cluster desc yaml when chips map is empty

pjanevskiTT, 2024-11-06:
As per offline chat with @abhullar-tt , I will take over the work on the branch she mentioned above

broskoTT, 2024-11-07:
@pjanevskiTT please raise a warning in code in this case
boardtype: {
   0: null,
}

I see that current bh CEM outputs that. We should probably resolve it, but not critical at the moment.

@TTDRosen any ideas why do we get this?

abhullar-tt, 2024-11-07:
It’s because some machines are running old FW

broskoTT, 2024-11-07:
Also, once this is resolved I think we can remove tt_ClusterDescriptor::create_for_grayskull_cluster

PR #280 (2024-11-09): Fix parsing cluster descriptor
### Issue

Fix #241 

### Description

Fix parsing of cluster descriptor yaml when there is no chips section (Blachkole and Grayskull path)

### List of the changes

- Fix parsing of cluster descriptor without chips section
- Add warning for unknown board type
- Remove `create_for_grayskull_cluster`
- Fix all tests to use `create-ethernet-map`
- Remove default fallback cluster descriptor for Blackhole

### Testing

Unit testing

### API Changes

No API changes

tt_debuda PR which removes CEM: https://github.com/tenstorrent/tt-debuda/pull/165
closed 2024-11-09 by pjanevskiTT