tenstorrent/tt-umd #251: delete called on non-final 'CoordinateManager' that has virtual functions but non-virtual destructor

nsmithtt, 2024-11-04:
tt-mlir's latest [uplift integration](https://github.com/tenstorrent/tt-mlir/actions/runs/11660244306/job/32462210096) fails because of the following compile error:

```c++
/usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/bits/unique_ptr.h:85:2: error: delete called on non-final 'CoordinateManager' that has virtual functions but non-virtual destructor [-Werror,-Wdelete-non-abstract-non-virtual-dtor]
   85 |         delete __ptr;
      |         ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/bits/unique_ptr.h:361:4: note: in instantiation of member function 'std::default_delete<CoordinateManager>::operator()' requested here
  361 |           get_deleter()(std::move(__ptr));
      |           ^
/__w/tt-mlir/tt-mlir/third_party/tt-metal/src/tt-metal/tt_metal/third_party/umd/device/tt_soc_descriptor.h:140:5: note: in instantiation of member function 'std::unique_ptr<CoordinateManager>::~unique_ptr' requested here
  140 |     tt_SocDescriptor(const tt_SocDescriptor& other) :
```

The fix should be to define a virtual destructor for class `CoordinateManager`.

nsmithtt, 2024-11-04:
fyi @pjanevskiTT

pjanevskiTT, 2024-11-04:
Looking at it

pjanevskiTT, 2024-11-04:
https://github.com/tenstorrent/tt-mlir/tree/pjanevski/fix_metal_uplift_build @nsmithtt let me know if the problem goes away

nsmithtt, 2024-11-04:
@pjanevskiTT, what is the uplift strategy when uplifting UMD into tt-metal?  We'd rather take vanilla main if we think tt-metal can adopt your fix in main.

pjanevskiTT, 2024-11-05:
Well we try to open the PR for every UMD change that requires API changes inside tt-metal, but merging that can sometimes take some time. Problems can occur with something like this that does not need to be uplifted in tt-metal, but you need the change. Do you have some suggestion on how we should do this?

nsmithtt, 2024-11-05:
Presumably UMD recently uplifted (within the past couple weeks) because we just started seeing this error.  Would it be possible to uplift again now?  Who can I contact regarding this process?

I think in this case we'll just be leaking memory.

pjanevskiTT, 2024-11-05:
I can uplift it. Let me create a branch with this commit included and run post-commit

nsmithtt, 2024-11-05:
That's ok with me, fyi @blozano-tt @afuller-TT @vtangTT @broskoTT

pjanevskiTT, 2024-11-05:
Closing the issue because it is fixed within umd, @nsmithtt I will tag all the people above in the tt-metal uplift

closed 2024-11-05 by pjanevskiTT