tenstorrent/tt-umd #257: Explore an IOMMU based scheme that mimics sysmem API for GS/WH

joelsmithTT, 2024-11-04:
In UMD, "sysmem" refers to 1G hugepages used as shared buffers between device and host.  The APIs for system include tt_SiliconDevice constructor (specifies number of "channels") and accessor methods (read/write/get pointer to buffer) that are channel based.

While preserving the API, write an alternate implementation that does not huge hugepages, but relies on the KMD ability to map physically noncontiguous buffers to a contiguous IO virtual address space seen by the device.

The challenges at the system level:
* WH hardware is limited to accessing an address space of 0x0-0xffff'ffff without iATU programming
* KMD IOVA allocation does not seem to support complete coverage of this address space
* With iATU we can work around this.  However, iATU is limited to 16 regions with a constraint that the IOVA must be naturally aligned with the region size.

Challenge at the user driver level:
* Maintain the illusion of one to four channels, even if there are up to 16 sub allocations

nsmithtt, 2024-11-04:
`dma_alloc_coherent` or `dma_pool_create` might be useful in getting aligned memory?

broskoTT, 2025-07-10:
I think that Joel already did a great job of mimicking current hugepage setup with iommu setup, we can close this one, it does not cover work currently being done in this field

closed 2025-07-10 by broskoTT