tenstorrent/tt-umd #275: [Bug Report] Huge page warning when 1st time running tt-metal after boot

marty1885, 2024-11-06:
**Describe the bug**
I just upgraded my local tt-metal today and found that the following is printed after updating API uses in my program. Then I find tt-metal print the following warning, which wasn't the case in the past. However, a second run of the program does not show the warning. And (after the 1st run, running `ls` shows the mappings does exist with exactly 1GB in size)

```
❯ bin/test-metalium
                 Device | INFO     | Opening user mode device driver
  Detecting chips (found 8)                                                                                                                          
2024-10-23 21:04:30.355 | INFO     | SiliconDriver   - Detected 4 PCI devices : [0, 1, 2, 3]
2024-10-23 21:04:30.403 | WARNING  | SiliconDriver   - Opened hugepage file has zero size, mapping it might fail: /dev/hugepages-1G/tenstorrent. Verify that enough hugepages are provided.
2024-10-23 21:04:30.457 | WARNING  | SiliconDriver   - Opened hugepage file has zero size, mapping it might fail: /dev/hugepages-1G/device_0_channel_1_tenstorrent. Verify that enough hugepages are provided.
2024-10-23 21:04:30.590 | INFO     | SiliconDriver   - Software version 6.0.0, Ethernet FW version 6.9.0 (Device 0)
2024-10-23 21:04:30.591 | INFO     | SiliconDriver   - Software version 6.0.0, Ethernet FW version 6.9.0 (Device 4)
2024-10-23 21:04:30.653 | INFO     | SiliconDriver   - Detected 4 PCI devices : [0, 1, 2, 3]
2024-10-23 21:04:30.656 | WARNING  | SiliconDriver   - Opened hugepage file has zero size, mapping it might fail: /dev/hugepages-1G/device_1_tenstorrent. Verify that enough hugepages are provided.
2024-10-23 21:04:30.709 | WARNING  | SiliconDriver   - Opened hugepage file has zero size, mapping it might fail: /dev/hugepages-1G/device_1_channel_1_tenstorrent. Verify that enough hugepages are provided.
2024-10-23 21:04:30.843 | INFO     | SiliconDriver   - Software version 6.0.0, Ethernet FW version 6.9.0 (Device 1)
2024-10-23 21:04:30.843 | INFO     | SiliconDriver   - Software version 6.0.0, Ethernet FW version 6.9.0 (Device 5)
2024-10-23 21:04:30.904 | INFO     | SiliconDriver   - Detected 4 PCI devices : [0, 1, 2, 3]
2024-10-23 21:04:30.908 | WARNING  | SiliconDriver   - Opened hugepage file has zero size, mapping it might fail: /dev/hugepages-1G/device_2_tenstorrent. Verify that enough hugepages are provided.
2024-10-23 21:04:30.961 | WARNING  | SiliconDriver   - Opened hugepage file has zero size, mapping it might fail: /dev/hugepages-1G/device_2_channel_1_tenstorrent. Verify that enough hugepages are provided.
2024-10-23 21:04:31.094 | INFO     | SiliconDriver   - Software version 6.0.0, Ethernet FW version 6.9.0 (Device 2)
2024-10-23 21:04:31.094 | INFO     | SiliconDriver   - Software version 6.0.0, Ethernet FW version 6.9.0 (Device 6)
2024-10-23 21:04:31.156 | INFO     | SiliconDriver   - Detected 4 PCI devices : [0, 1, 2, 3]
2024-10-23 21:04:31.160 | WARNING  | SiliconDriver   - Opened hugepage file has zero size, mapping it might fail: /dev/hugepages-1G/device_3_tenstorrent. Verify that enough hugepages are provided.
2024-10-23 21:04:31.212 | WARNING  | SiliconDriver   - Opened hugepage file has zero size, mapping it might fail: /dev/hugepages-1G/device_3_channel_1_tenstorrent. Verify that enough hugepages are provided.
2024-10-23 21:04:31.346 | INFO     | SiliconDriver   - Software version 6.0.0, Ethernet FW version 6.9.0 (Device 3)
2024-10-23 21:04:31.347 | INFO     | SiliconDriver   - Software version 6.0.0, Ethernet FW version 6.9.0 (Device 7)
                  Metal | INFO     | Initializing device 0. Program cache is NOT enabled
                  Metal | INFO     | AI CLK for device 0 is:   1000 MHz

```

**To Reproduce**
Steps to reproduce the behavior:
1. Compile tt-metal version `ad7a012c3fd79c70cf3d298949944f31fda30fa8` and one of the TTNN example programs
2. Reboot the machine
3. Run one of the example programs
4. Observe the error

**Expected behavior**
The warnings should not be printed

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Please complete the following environment information:**
- OS: Arch Linux, kernel 6.6.57
- tt-metal: ad7a012c3fd79c70cf3d298949944f31fda30fa8
- tt-kmd: 1.29
- CPU: AMD EPYC 8124P
- Card: Wormhole N300 x4

**Additional context**
Add any other context about the problem here.

broskoTT, 2024-11-06:
This warning is by design. I didn't want to leave the state so that if the following bug is hit again somebody loses as much time as I did investigating: https://github.com/tenstorrent/tt-umd/pull/168 and https://github.com/tenstorrent/tt-metal/issues/13618.

I realise that it brings confusion now that it is printed even when the system is healthy. Will change it so that it is only printed if hugepage mapping fails.

In the meantime, if nothing fails and this warning is printed, it is safe to ignore it.

broskoTT, 2024-11-07:
We're doing many changes in the API between tt_metal and tt_umd, so we're updating umd rather frequently. Some of those PRs will pick up this change from UMD. So no additional action is needed here after the linked PR is merged.

closed 2024-11-07 by broskoTT