tenstorrent/tt-umd #309: Tests are not relocatable

blozano-tt, 2024-11-16:
I am having some problem with UMD tests.
If UMD is built on machine A. (Highly parallel builder machine with no Tenstorrent cards).
And I run tests on machine B. (build products transferred to arbitrary directory on machine B from machine A).

I end up hitting this error:
```
unknown file: Failure
C++ exception with description "Error: device descriptor file /home/ubuntu/actions-runner-2/_work/tt-metal/tt-metal/tt_metal/third_party/umd/tests/soc_descs/grayskull_10x12.yaml does not exist!" thrown in the test body.
```

The absolute path from the builder machine is baked into the test case! I don't have this path on machine B.
Running tests from the build tree is usually a problem like this, this is why there is the concept of installing tests, libraries, meta data files. Artifacts in an install tree are usually relocatable. i.e. they can be in /usr/lib or /opt/tenstorrent or what have you, and everything still works.
How can we make this better?
ENV that points to the yaml?
Embed the yaml info into the library?
Start running tests from the install tree? (edited)

broskoTT, 2024-11-16:
Yes this is a problem introduced in https://github.com/tenstorrent/tt-umd/pull/302.
What is the usecase during which you are hitting this issue? This was also true before for tests using soc descriptors, which are arch specific tests.

blozano-tt, 2024-11-16:
> Yes this is a problem introduced in [#302](https://github.com/tenstorrent/tt-umd/pull/302). What is the usecase during which you are hitting this issue? This was also true before for tests using soc descriptors, which are arch specific tests.

I was trying to optimize the “all post commit” pipeline in tt-metal.
For most things we build on our builder machines and we test on VMs with the required hardware.
UMD was the only job where we built and tested on VMs.
Building on VMs is slow, and in this case it should be redundant because we have a build stage that already builds everything.
I thought I could reduce the burden on our VMs if I made UMD conform to the strategy of build on machine A, test on machine B.

broskoTT, 2025-07-21:
This is not a problem anymore and it is covered by CI.
We don't use yaml files in tests.