tenstorrent/tt-umd #315: Migrate log statements to new logger

joelsmithTT, 2024-11-18:
* Rename logger_.cpp, logger_.hpp to logger.cpp, logger.hpp; remove old versions of these files
* Either rename the macros in new logger or rename the macros currently in use (or both...).
* Figure out what to do with the `LogSiliconDriver` in current log messages.  It's not used consistently (log_error and log_assert do not have it).  Suggestion is to just remove it it for now and have a single logger.
* Get rid of log_assert, or at least make it do what its name says.  If we want it to throw an exception, it should have the word "throw" in it rather than assert.
* Fix whatever compile errors result due to broken log messages being exposed

broskoTT, 2025-07-21:
I think other issues cover most of the leftover tasks mentioned here.
Some of those were naturally closed by the introduction of tt-logger

broskoTT, 2025-08-27:
Yeah I think it was done by other PRs... Mainly by https://github.com/tenstorrent/tt-umd/pull/883

closed 2025-08-27 by broskoTT