tenstorrent/tt-umd #356: Coding style

broskoTT, 2024-11-30:
Make a brief document on the coding style: 
- Namings of classes, functions, private members, etc
- File structure
- Usage of namesspace
- Potentially some dos and donts

broskoTT, 2024-11-30:
+ whether header files are .h or .hpp

broskoTT, 2024-11-30:
+ How are arguments passed to a function. Whether it is a reference or pointer if it will be modified. Whether the primitive types are also passed as const references. 
+ reference google style guide

@blozano-tt @afuller-TT , just wondering if there is something already in place which answers these questions for tt_metal.

blozano-tt, 2024-11-30:
> * How are arguments passed to a function. Whether it is a reference or pointer if it will be modified. Whether the primitive types are also passed as const references.
> * reference google style [...]

I'm not aware of anything that focuses on style, just some best practices.

Metal uses a mix of .h and .hpp for C++ header files.

There are some documents here that dictate some general best practices, like "don't use bool function arguments.

https://github.com/tenstorrent/tt-metal/tree/main/contributing

broskoTT, 2024-12-23:
- using `auto` keyword

blozano-tt, 2024-12-23:
> * using `auto` keyword

I love auto

broskoTT, 2024-12-31:
- mutable keyword (when changing some field in a const reference to an object.

blozano-tt, 2024-12-31:
> * mutable keyword (when changing some field in a const reference to an object.

This always felt like a hack to me. Is there a legit reason to do this?

broskoTT, 2025-01-03:
Just to be clear, the points I raise in this issue is just the points that I want to be addressed with coding style document, I'm still not suggesting anything yet.

For mutable: when you want a field to be modifiable in a const object. Yes, no legit reason to use it.
Actually, just googled it. One potentially legit usage is for internal mutex, so that you can call a .read() const method which need some synchronization.

When you create yourself a "const hell" by having a lot of consts wherever it was possible, this might be the only option sometimes.

afuller-TT, 2025-01-06:
> > * mutable keyword (when changing some field in a const reference to an object.
> 
> This always felt like a hack to me. Is there a legit reason to do this?

`std::mutex`.  A getter should be const, but the mutex must be non-const to guard against a setting joining the party.

In general `mutable` deserves a lot of extra scrutiny.  It should be rare.  But it does have its place.

broskoTT, 2025-01-22:
- Tests should not have any programming/ any logic done to the values. Use hardcoded values for comparison. Otherwise you end up testing with the same code that is in the original code.

broskoTT, 2025-03-24:
- usage of "or" and "and" in c++

broskoTT, 2025-03-25:
- All structs should have .str() implementation
- Investigate .count() and .contains() usage instead of .find() != .end()

broskoTT, 2025-03-25:
- How to write tests and not just reimplemented the tested code. Possible definition: When you have an EXPECT statement, you should not have calculated values on both sides. One of those values has to be a hardcoded constant, a referenced constant, or derived from constants but only with simple math operations.

broskoTT, 2025-03-31:
- Destructor should be right after the constructor (position of the destructor in header)

broskoTT, 2025-05-22:
- "" instead of <> for includes

broskoTT, 2025-06-09:
- uint32_t vs size_t, but also other _t types

blozano-tt, 2025-06-09:
> * uint32_t vs size_t, but also other _t types

https://google.github.io/styleguide/cppguide.html#Integer_Types

broskoTT, 2026-09-12:
I've put a lot of ideas here for somebody eventually to go over them and write our style guide.

The scope of this task includes going through all the suggestions mentioned in this issue
The goal of the task is to produce a coding style guide for developers, and/or enforce some of these rules through clang-tidy

moklobdzijaTT, 2026-10-01:
Take into consideration the to-be-introduced agent directives, that they should be also updated/reorganized when coding style is defined. See #3294 for more info.

broskoTT, 2026-11-04:
Triage note: partial progress since this was opened.
- Test-writing rules (hardcoded expected values), commenting and logging rules are now in `AGENTS.md` (#3294).
- `.h` -> `.hpp` settled by #1219.

Still open: naming conventions (`readability-identifier-naming` has no `CheckOptions` in `.clang-tidy`, so it enforces nothing), argument passing, `auto`, `mutable`, protected data, include quoting, integer types, hex separators, brace init, docs naming/structure. Suggest the deliverable be `docs/CODING_STYLE.md` linked from `AGENTS.md`, with each rule either enforced by `.clang-tidy`/`.clang-format` or explicitly marked as review-only.