tenstorrent/tt-umd #363: RandomSysmemTestWithPcie failing on tt_metal CI

broskoTT, 2024-12-03:
Doesn't seem like this is happening on a single VM though

tt-metal-ci-vm-95
https://github.com/tenstorrent/tt-metal/actions/runs/12132413898/job/33826246222

tt-metal-ci-vm-39
https://github.com/tenstorrent/tt-metal/actions/runs/12132649762/job/33826853456
https://github.com/tenstorrent/tt-metal/actions/runs/12132909850/job/33827560669

tt-metal-ci-vm-98
https://github.com/tenstorrent/tt-metal/actions/runs/12129290035/job/33817421580

joelsmithTT, 2024-12-03:
This is failing because the environment is broken.  This is a Metal infra bug.

broskoTT, 2024-12-04:
Caused by https://github.com/tenstorrent/tt-metal/issues/15675

broskoTT, 2025-01-28:
Not happening anymore in tt-metal, because umd tests were disabled. This issue was also somewhat fixed in tt-metal.

closed 2025-01-28 by broskoTT