tenstorrent/tt-umd #368: Cluster::read/write_mmio_device_register dynamic tlb issue

lgojic-tt, 2024-12-04:
https://github.com/tenstorrent/tt-umd/blob/48cae2bec98c6a87441407221bc88046d3cc893f/device/cluster.cpp#L3037-L3055

Read/write_mmio_device_register sets up only one dynamic tlb when reading/writing. This is problematic because when reading more than 16 mb (the size of the dynamic tlb), the function will wrap around after 16mb. In this example I have written the address into the every value in the first 32mb. Do note that I have done this using the LARGE_WRITE_TLB, which uses a different function called "write_to_device_memory" which does set more than one dynamic tlb if there is a need for it.

Reading 32mb from the address 0x0

[image]

As we can see after the first 16mb, the values wrap around to the beginning. 
The same thing happens at the border of 16mb if we read from the address 0x00ffff00. 

The fix would be to do the same thing as in read_device_memory, set the dynamic tlb as many times as its needed for the read to read everything. 

https://github.com/tenstorrent/tt-umd/blob/48cae2bec98c6a87441407221bc88046d3cc893f/device/cluster.cpp#L1198-L1211

If this is not possible because of some design choice, at least an assert needs to be placed to notify the user that they cannot read more than 16mb, or read memory from 16mb border(or 32mb border etc..). The same is true for write_mmio_device register.

broskoTT, 2025-03-10:
We need to refactor read/write device to separate register IO from other access: #596

broskoTT, 2025-04-03:
We're currently in a process of rewriting this code. We should recheck if this is still the issue then.
It also might've been fixed already by https://github.com/tenstorrent/tt-umd/issues/281.