tenstorrent/tt-umd #398: Add ability to read/write noc0 addresses that are not in SoC descriptor

tt-vjovanovic, 2024-12-12:
At the moment, not all components are stored in SoC descriptor and for debugging scenarios we might need to read/write to those components... CoordinateManager doesn't need to translate those addresses to other coordinate systems (using `noc0`/`noc1` is enough).

broskoTT, 2025-03-10:
Which components are not stored in SoC descriptor? Everything that the driver should be aware about should be present in SoC, right? otherwise we have a "parallel SoC" in UMD. Or do you mean that if tt-lens requests reading from some noc address not present in SOC umd should not block it even if it doesn't know what would be there?

tt-vjovanovic, 2025-03-10:
Everything that UMD should know about should be part of SoC descriptor. Just add "feature" so UMD doesn't block even if it thinks there isn't something there. Something like superuser ability...
Basically, if case there is something we are still not supporting with UMD tt-exalens should be able to get to it since it is debugging utility...

broskoTT, 2025-03-10:
Ok so let's say that could be a "warning". We just have to figure out what to do with coordinates which we would try to translate...

pjanevskiTT, 2025-07-22:
Yes we exposed all core types. Some core types with multiple NOC ports (l2cpu for example) have some NOC ports storted in router, but there is at least one NOC port for each NOC tile exposed. If this is something that is enough for tt-exalens at the moment we can close the issue @broskoTT @tt-vjovanovic