tenstorrent/tt-umd #399: Add support for read/write timeout

tt-vjovanovic, 2024-12-12:
When #398 is implemented, we power user that is using that feature might make a mistake. In order not to require application restart cause UMD will get stuck, we should consider adding timeout for noc transactions that don't return result (cause endpoint doesn't exist or is not responding).

Also, if UMD is using wrong ETH core to read/write something from/to remote chip, we should also timeout...

joelsmithTT, 2024-12-19:
Related: https://github.com/tenstorrent/tt-umd/issues/291

pjanevskiTT, 2025-02-13:
@tt-vjovanovic can you describe the issue we talked about reading memory through hardware debug interface in a bit more detail? How does NOC behave after you see the hang? You can link the issue on tt-lens side if there is some

tt-vjovanovic, 2025-02-25:
Is this issue ok?
https://tenstorrent.atlassian.net/browse/TEN-1919

broskoTT, 2025-09-19:
This has also been requested as part of #1348

pjanevskiTT, 2025-10-23:
MMIO reads/writes are protected by timeout on PCIe controller, since remote timeout is implemented I will close the issue