tenstorrent/tt-umd #431: Metal SW requests for UMD cluster descriptor

aliuTT, 2024-12-20:


abhullar-tt, 2025-01-17:
I think we need to have one API return a collection of active ethernets (includes links that are down and successfully trained) and then another API to check if the link is active

pjanevskiTT, 2025-01-27:
@cfjchu can you write an example how would you like yaml created by cluster descriptor to look like? You can create separate issue as well. For example p300 cluster descriptor, I guess everything can be represented there

cfjchu, 2025-01-30:
@pjanevskiTT sorry, I missed this message and @abhullar-tt pinged me about it.

From what I see, tt-metal interfaces with UMD through:
```
1. class umd::Cluster : public umd::tt_device
2. tt_ClusterDescriptor
```

Here's my use-case:

1. On a machine with a silicon device, I'd like to be able to `Cluster::serialize()` to some yaml 
2. On a machine without a silicon device, I want to fake/virtualize the `tt_device` using something like `tt_MockupDevice` or `tt_SimulationDevice` and construct it using that serialized yaml. It should be able to service queries like `get_ethernet_connections`, which become static lookups into some descriptor object

adifrancescoTT, 2025-02-28:
Checking if there is an ETA for this? Thanks

closed 2025-03-06 by broskoTT