tenstorrent/tt-umd #461: Compiler error in Anaconda with gcc 14

blozano-tt, 2025-01-18:
Hitting a strange compiler error in UMD when building inside of an Anaconda environment, with gcc 12.4
sys/types.h seems to be providing a macro that shadows our symbols in symver_t
horrible name to give a macro if this is real

```
akeFiles/device.dir/wormhole/wormhole_implementation.cpp.o -c $SRC_DIR/tt_metal/third_party/umd/device/wormhole/wormhole_implementation.cpp
In file included from $BUILD_PREFIX/x86_64-conda-linux-gnu/sysroot/usr/include/sys/types.h:222,
                 from $BUILD_PREFIX/x86_64-conda-linux-gnu/sysroot/usr/include/stdlib.h:314,
                 from $BUILD_PREFIX/lib/gcc/x86_64-conda-linux-gnu/12.4.0/include/c++/cstdlib:75,
                 from $BUILD_PREFIX/lib/gcc/x86_64-conda-linux-gnu/12.4.0/include/c++/ext/string_conversions.h:41,
                 from $BUILD_PREFIX/lib/gcc/x86_64-conda-linux-gnu/12.4.0/include/c++/bits/basic_string.h:3972,
                 from $BUILD_PREFIX/lib/gcc/x86_64-conda-linux-gnu/12.4.0/include/c++/string:53,
                 from $BUILD_PREFIX/lib/gcc/x86_64-conda-linux-gnu/12.4.0/include/c++/bitset:47,
                 from $BUILD_PREFIX/lib/gcc/x86_64-conda-linux-gnu/12.4.0/include/c++/regex:38,
                 from $SRC_DIR/tt_metal/third_party/umd/device/api/umd/device/tt_xy_pair.h:9,
                 from $SRC_DIR/tt_metal/third_party/umd/device/api/umd/device/architecture_implementation.h:14,
                 from $SRC_DIR/tt_metal/third_party/umd/device/api/umd/device/wormhole_implementation.h:11,
                 from $SRC_DIR/tt_metal/third_party/umd/device/wormhole/wormhole_implementation.cpp:5:
$SRC_DIR/tt_metal/third_party/umd/device/api/umd/device/semver.hpp: In constructor 'tt::umd::semver_t::semver_t(uint64_t, uint64_t, uint64_t)':
$SRC_DIR/tt_metal/third_party/umd/device/api/umd/device/semver.hpp:28:64: error: class 'tt::umd::semver_t' does not have any field named 'gnu_dev_major'
   28 |     semver_t(uint64_t major, uint64_t minor, uint64_t patch) : major(major), minor(minor), patch(patch) {}
      |                                                                ^~~~~
$SRC_DIR/tt_metal/third_party/umd/device/api/umd/device/semver.hpp:28:78: error: class 'tt::umd::semver_t' does not have any field named 'gnu_dev_minor'
   28 |     semver_t(uint64_t major, uint64_t minor, uint64_t patch) : major(major), minor(minor), patch(patch) {}
      |                                                                              ^~~~~
In file included from $SRC_DIR/tt_metal/third_party/umd/device/api/umd/device/pci_device.hpp:16,
                 from $SRC_DIR/tt_metal/third_party/umd/device/api/umd/device/tt_device/tt_device.h:10,
                 from $SRC_DIR/tt_metal/third_party/umd/device/api/umd/device/cluster.h:22,
                 from $SRC_DIR/tt_metal/third_party/umd/device/wormhole/wormhole_implementation.cpp:7:
$SRC_DIR/tt_metal/third_party/umd/device/api/umd/device/semver.hpp:28:5: error: uninitialized const member in 'const uint64_t' {aka 'const long unsigned int'} [-fpermissive]
   28 |     semver_t(uint64_t major, uint64_t minor, uint64_t patch) : major(major), minor(minor), patch(patch) {}
      |     ^~~~~~~~
$SRC_DIR/tt_metal/third_party/umd/device/api/umd/device/semver.hpp:24:20: note: 'const uint64_t tt::umd::semver_t::major' should be initialized
   24 |     const uint64_t major;
```

broskoTT, 2025-04-03:
low priority, as we don't build for this configuration. If this is used somewhere in some development workflow, let us know to bump the priority

broskoTT, 2025-07-09:
We should probably rename semver_t to avoid these in the future.
Somewhat related to #354

broskoTT, 2025-07-10:
Happy to close it then :) Thanks