tenstorrent/tt-umd #465: Device resource reservations

pjanevskiTT, 2025-01-20:
Device resource reservation is going to be implemented in https://github.com/tenstorrent/tt-kmd. The point of this feature is to enable two separate workloads to happen on a chip, or enable some utility things to happen on a chip without interfering with workload (X280 use case)

API for resource reservation is going to be exposed to UMD using few ioctl calls

```
1. RegisterReservation (Type, arg0, arg1, ....) => reservation id. 
Based on the reservation type, multiple arguments might be important. For example, for tensix reservations we would like to have starting point and grid size, for dram banks, ....
2. Release reservation: reservation id => (no return value, reservation is released) 
3. List all reservations: ReservationType => List of all reservations```
Issue with more details on device resource reservation in  https://github.com/tenstorrent/tt-kmd -> TODO: add KMD issue
```

Issue with more details on the design in https://github.com/tenstorrent/tt-kmd -> TODO: add issue in KMD

broskoTT, 2025-01-20:
[image]

broskoTT, 2026-11-04:
Deprecated in favour of #3811. The KMD reservation ioctls proposed here were never implemented, and KMD resource locks (`LOCK_CTL`) now cover cross-process mutual exclusion. The remaining need, visibility and ownership of device resources (DRISCs used by FW, tt-metal and tt-dgen/tt-inference, plus debug tools like exalens; see #2531), is tracked in #3811.

closed 2026-11-04 by broskoTT