tenstorrent/tt-umd #491: Create cluster description for simulation device

tt-vjovanovic, 2025-01-29:
Currently, we have mechanism to to detect cluster description with `tt_ClusterDescriptor::get_cluster_descriptor_file_path()` for devices that are attached to host. We also have `tt_ClusterDescriptor::create_mock_cluster` function that creates mock version for specified device ids and architecture.

Minimum what we should have is function that will create cluster descriptor for simulation device based on simulator directory.

Since we will have function for "real" devices connected to host and also function for simulaton device, not sure if it makes sence to have static methods in `tt::umd::Cluster` and `tt_SimulationDevice` that would be logically associated with those functions...

broskoTT, 2025-07-21:
isn't create_mock_cluster already that? It is just not getting the arch name from the folder it was passed, but other than that it should be it, right?
Can you elaborate?

broskoTT, 2025-09-15:
I don't understand the full impact of this. Maybe we do want to have a silicon board type for simulator?
Are you referring to both arch and board_type, or just boardtype?

pjanevskiTT, 2026-10-01:
TTSim clusters now get their cluster descriptor from `TopologyDiscovery` rather than a supplied file (`Cluster`, `device/cluster.cpp`). The remaining gaps (RTL, Quasar, remote chips) are tracked in #3353. Closing.

closed 2026-10-01 by pjanevskiTT