tenstorrent/tt-umd #505: BH unit tests failing

broskoTT, 2025-02-05:
Also, we don't have a BH machine in CI, this should be added in anycase.

`[ RUN      ] SiliconDriverBH.DynamicTLB_RW
2025-02-05 08:14:58.021 | INFO     | SiliconDriver   - Opened PCI device 0; KMD version: 1.29.0, IOMMU: disabled
2025-02-05 08:14:58.229 | INFO     | SiliconDriver   - Detected PCI devices: [0]
2025-02-05 08:14:58.229 | INFO     | SiliconDriver   - Using local chip ids: {0} and remote chip ids {}
2025-02-05 08:14:58.231 | INFO     | SiliconDriver   - Device: 0 Mapped iATU region 0 from 0x0 to 0x3fffffff to 0x180000000
Target Tensix cores completed
/home/brosko/tt-umd/tests/blackhole/test_cluster_bh.cpp:459: Failure
Expected equality of these values:
  vector_to_write
    Which is: { 0, 1, 2, 3, 4, 5, 6, 7, 8, 9 }
  readback_vec
    Which is: { 4294901760, 4294901761, 2, 3, 4, 5, 6, 7, 8, 9 }
Vector read back from core 0-9does not match what was written
[  FAILED  ] SiliconDriverBH.DynamicTLB_RW (841 ms)
[ RUN      ] SiliconDriverBH.MultiThreadedDevice
2025-02-05 08:14:58.864 | INFO     | SiliconDriver   - Opened PCI device 0; KMD version: 1.29.0, IOMMU: disabled
2025-02-05 08:14:59.072 | INFO     | SiliconDriver   - Detected PCI devices: [0]
2025-02-05 08:14:59.072 | INFO     | SiliconDriver   - Using local chip ids: {0} and remote chip ids {}
2025-02-05 08:14:59.074 | INFO     | SiliconDriver   - Device: 0 Mapped iATU region 0 from 0x0 to 0x3fffffff to 0x180000000
/home/brosko/tt-umd/tests/blackhole/test_cluster_bh.cpp:530: Failure
Expected equality of these values:
  vector_to_write
    Which is: { 0, 1, 2, 3, 4, 5, 6, 7, 8, 9 }
  readback_vec
    Which is: { 0, 1, 4294311938, 4294770691, 4294901764, 4294901765, 4290707462, 3740205063, 8, 9 }
Vector read back from core 0-4does not match what was written
[  FAILED  ] SiliconDriverBH.MultiThreadedDevice (697 ms)`

broskoTT, 2025-02-05:
This could be a single machine failure, we should confirm this on more cards.

smehtaTT, 2025-02-19:
FYI @joelsmithTT and @broskoTT - https://github.com/tenstorrent/tt-umd/pull/527

broskoTT, 2025-02-26:
We now have a CI in place.
And these failures were observed on known bad bh machines in IRD. 
Closing this issue

closed 2025-02-26 by broskoTT