tenstorrent/tt-umd #522: get rid of / make optional: flatbuffers

alex-s168, 2025-02-07:
Flatbuffers is a heavy dependency and takes up a major amount of build time.
This is not important, but it would be more convenient if flatbuffers was optional or removed completely
It seems like it is only used in [tt_simulation_device.fbs](https://github.com/tenstorrent/tt-umd/blob/3e6e65629ad41d8c38ec20138964bd2a5b399187/device/simulation/tt_simulation_device.fbs)

pjanevskiTT, 2025-02-24:
@alex-s168 thanks for opening this issue

I also think this just makes the header file with same structs as in flatbuffer, looks like it can be moved to .h file.

Have you measured the exact time of the build time difference when we use .fbs and when this is moved to the .h file? Anyway, It would be nice to remove it no matter how big the build time difference is. 

If you want to work on it, you can assign it yourself or I can assign it to you.

alex-s168, 2025-04-09:
@pjanevskiTT 
I'll start working on this now.
Can you assign me please

broskoTT, 2025-04-23:
Fixed by #728

closed 2025-04-23 by broskoTT