tenstorrent/tt-umd #523: Proper synchronization on TTDevice read/writes

pjanevskiTT, 2025-02-13:


broskoTT, 2025-03-10:
@pjanevskiTT More info here?

pjanevskiTT, 2025-03-11:
There was a bug with mutex in TTDevice, where we didn't clean it up properly so #524 was checked in to unblock metal. Basically issue here is just to add proper creation/cleaning of mutex in TTDevice. I am kinda hoping that this goes away by KMD TLB implementation, but if not we should take care of this just by mutex implementation, similar to how it's done in Cluster today.

pjanevskiTT, 2025-03-11:
This is probably going to be needed for #555 since part of the topology discovery for both WH and BH is done through TTDevice

pjanevskiTT, 2025-03-11:
Assigning @broskoTT since this may be on critical path for #555

broskoTT, 2025-04-09:
@pjanevskiTT can we close this now that locks are merged?

pjanevskiTT, 2025-04-09:
yes, closing it

closed 2025-04-09 by pjanevskiTT