tenstorrent/tt-umd #525: Setup a CI BH machine for tt-umd

vmilosevic, 2025-02-18:
# Setup a CI Blackhole machine for tt-umd project.

- install docker
- check if FW is installed and if tt-smi works
- install action-runner
- register machine with tt-umd
- setup reset scripts

## Details

Michael Chiou:
This is a CI machine I just ripped out of metal CI for now. Comes with
```
TTKMD: 1.31
fw: 80.14.2.0
ssh ubuntu@172.27.44.195
```
Try using this machine for now.
In the near future, I may need to swap this VM with another one.
at that time, I will provide you a new one (fresh instance) and let you set that up and then take back the old VM when the new VM is done
It’s a P150

vmilosevic, 2025-02-18:
docker is installed
tt-smi-metal works

[image]

vmilosevic, 2025-02-18:
Registered for tt-umd as `umd-p150-1`. You can try it in CI with labels `p150` or `blackhole`
@broskoTT @pjanevskiTT

pjanevskiTT, 2025-02-24:
@broskoTT should be closed by #527 ?

closed 2025-02-25 by vmilosevic