tenstorrent/tt-umd #528: UMD Unit Tests Hang on T3K Systems

blozano-tt, 2025-02-20:
Michael and I are seeing UMD unit tests hang on two of our T3K baremetal CI systems.

Commit:
```
$ git log 
commit ebb0f945ed8d3c05e043158978201ed6fab884ec (grafted, HEAD)
Author: pjanevski <pjanevski@tenstorrent.com>
Date:   Sun Feb 9 01:05:46 2025 +0000

    Remove TTDevice mutex
```

```
$ ./build/test/umd/wormhole_b0/unit_tests
Running main() from /home/ubuntu/actions-runner/_work/tt-metal/tt-metal/.cpmcache/googletest/96129d89f45386492ae46d6bb8c027bc3df5f949/googletest/src/gtest_main.cc
[==========] Running 20 tests from 2 test suites.
[----------] Global test environment set-up.
[----------] 14 tests from SiliconDriverWH
[ RUN      ] SiliconDriverWH.CreateDestroy
  Detecting chips (found 8)                                                                                                                                                                                                                          
2025-02-20 19:35:19.946 | INFO     | SiliconDriver   - Skipping harvesting for chip 0.
2025-02-20 19:35:19.946 | INFO     | SiliconDriver   - Skipping DRAM harvesting for chip 0.
2025-02-20 19:35:19.946 | INFO     | SiliconDriver   - Skipping ETH harvesting for chip 0.
2025-02-20 19:35:19.948 | INFO     | SiliconDriver   - Opened PCI device 0; KMD version: 1.29.0, IOMMU: disabled
2025-02-20 19:35:19.948 | INFO     | SiliconDriver   - Skipping harvesting for chip 1.
2025-02-20 19:35:19.948 | INFO     | SiliconDriver   - Skipping DRAM harvesting for chip 1.
2025-02-20 19:35:19.948 | INFO     | SiliconDriver   - Skipping ETH harvesting for chip 1.
2025-02-20 19:35:19.949 | INFO     | SiliconDriver   - Opened PCI device 1; KMD version: 1.29.0, IOMMU: disabled
2025-02-20 19:35:19.950 | INFO     | SiliconDriver   - Skipping harvesting for chip 2.
2025-02-20 19:35:19.950 | INFO     | SiliconDriver   - Skipping DRAM harvesting for chip 2.
2025-02-20 19:35:19.950 | INFO     | SiliconDriver   - Skipping ETH harvesting for chip 2.
2025-02-20 19:35:19.951 | INFO     | SiliconDriver   - Opened PCI device 2; KMD version: 1.29.0, IOMMU: disabled
2025-02-20 19:35:19.951 | INFO     | SiliconDriver   - Skipping harvesting for chip 3.
2025-02-20 19:35:19.951 | INFO     | SiliconDriver   - Skipping DRAM harvesting for chip 3.
2025-02-20 19:35:19.951 | INFO     | SiliconDriver   - Skipping ETH harvesting for chip 3.
2025-02-20 19:35:19.952 | INFO     | SiliconDriver   - Opened PCI device 3; KMD version: 1.29.0, IOMMU: disabled
2025-02-20 19:35:19.953 | INFO     | SiliconDriver   - Skipping harvesting for chip 4.
2025-02-20 19:35:19.953 | INFO     | SiliconDriver   - Skipping DRAM harvesting for chip 4.
2025-02-20 19:35:19.953 | INFO     | SiliconDriver   - Skipping ETH harvesting for chip 4.
2025-02-20 19:35:19.954 | INFO     | SiliconDriver   - Skipping harvesting for chip 5.
2025-02-20 19:35:19.954 | INFO     | SiliconDriver   - Skipping DRAM harvesting for chip 5.
2025-02-20 19:35:19.954 | INFO     | SiliconDriver   - Skipping ETH harvesting for chip 5.
2025-02-20 19:35:19.954 | INFO     | SiliconDriver   - Skipping harvesting for chip 6.
2025-02-20 19:35:19.954 | INFO     | SiliconDriver   - Skipping DRAM harvesting for chip 6.
2025-02-20 19:35:19.954 | INFO     | SiliconDriver   - Skipping ETH harvesting for chip 6.
2025-02-20 19:35:19.955 | INFO     | SiliconDriver   - Skipping harvesting for chip 7.
2025-02-20 19:35:19.955 | INFO     | SiliconDriver   - Skipping DRAM harvesting for chip 7.
2025-02-20 19:35:19.955 | INFO     | SiliconDriver   - Skipping ETH harvesting for chip 7.
2025-02-20 19:35:19.956 | INFO     | SiliconDriver   - Detected PCI devices: [0, 1, 2, 3]
2025-02-20 19:35:19.956 | INFO     | SiliconDriver   - Using local chip ids: {0, 1, 2, 3} and remote chip ids {4, 5, 6, 7}
2025-02-20 19:35:20.059 | INFO     | SiliconDriver   - Software version 6.0.0, Ethernet FW version 6.10.0 (Device 0)
2025-02-20 19:35:20.059 | INFO     | SiliconDriver   - Software version 6.0.0, Ethernet FW version 6.10.0 (Device 1)
2025-02-20 19:35:20.059 | INFO     | SiliconDriver   - Software version 6.0.0, Ethernet FW version 6.10.0 (Device 2)
2025-02-20 19:35:20.059 | INFO     | SiliconDriver   - Software version 6.0.0, Ethernet FW version 6.10.0 (Device 3)
2025-02-20 19:35:20.060 | INFO     | SiliconDriver   - Software version 6.0.0, Ethernet FW version 6.10.0 (Device 4)
```

ttmchiou, 2025-02-20:
note, there is also a working theory that this could be due to hardware issues and not a umd issue per se. Will update more later

broskoTT, 2025-02-26:
@joelsmithTT  made a fix https://github.com/tenstorrent/tt-umd/pull/533
Not sure if this is all there is to it, someone should confirm if this is still happening.
Note: Longer term, UMD should have all supported configurations in CI. I hope that once cloud team starts releasing shared pools of various systems we could use those pools to verify UMD is working

blozano-tt, 2025-02-26:
The hang is believed to happen when the chip is in a bad state. A reset is believed to be the cure.
It would be good to understand why this hang is possible, but I dropped this down to P2.

closed 2025-03-05 by joelsmithTT