tenstorrent/tt-umd #555: Support multiple UMD processes

broskoTT, 2025-02-28:
After syncing with tt-fabric team, we clarified their requirements from us.

The tools (which use Luwen) running alongside ai workload (through tt-metal and umd) is not a requirement.

The scenario is as follows: 
There will be a health check process long running in the background, which will be built somewhere in stack, but on top of UMD. (TBD: Thing whether Cluster or TTDevice is a better entry point for this).
Alongside this process, user will turn workload on and off, e.g. another UMD process will spawn up on the same set of chips.

These two should work without stepping on each other's toes. Making this functional anyway possible is a priority. Complete security is of lower priority, since clients are assumed not to be malicious. For example, if both processes start programming same TLBs, that would crash the device or workload, and that is not guaranteed with this implementation.

Further work is planned, in #462 which has this goal, so we don't need another implementation for that in the meantime.

As Aditya pointed out, currently multiple UMD processes are possible, but only if one process creates and initialises UMD driver and then forks several processes from it. This doesn't exactly match the scenario requested though.

A proposal is to make a test, or a set of tests, spawning multiple UMD processes and stress testing communication with the chip/chips and verifying nothing crashes. And then doing whatever needed to make it work.

FYI @pjanevskiTT 
Please let me know if I misunderstood your scenario or requirements, FYI @tt-asaigal @abhullar-tt @aliuTT

pjanevskiTT, 2025-02-28:
I want to add that supporting #523 is probably going to be part of this effort if we want two UMD processes doing the topology discover in parallel

pjanevskiTT, 2025-03-24:
TODO: address code organization comments https://github.com/tenstorrent/tt-umd/pull/624#discussion_r2009278971 when Cluster mutexes are consolidated to be used together with TTDevice and ArcMessenger

pjanevskiTT, 2025-03-31:
@broskoTT can we close this as feature done and focus on bugs in #654 ?

pjanevskiTT, 2025-04-11:
closed by #647

closed 2025-04-11 by pjanevskiTT