tenstorrent/tt-umd #558: Scripts to Automate Installing Dependencies, Building, and Running Tests for tt-umd

kkabilarTT, 2025-02-28:
Have 3 folders (one for grayskull, wormhole, and blackhole) each with 3 scripts: installing dependencies, building, and running test

[tt-umd-blackhole.zip](https://github.com/user-attachments/files/19030419/tt-umd-blackhole.zip)

[tt-umd-grayskull.zip](https://github.com/user-attachments/files/19030420/tt-umd-grayskull.zip)

[tt-umd-wormhole.zip](https://github.com/user-attachments/files/19030422/tt-umd-wormhole.zip)

bkeith-TT, 2025-02-28:
Some more context for this:

The goal is to 'shift-left' and provide scripts so that those upstream from Metal (Firmware & Cloud teams) could run these tests to test hardware, firmware, and hardware/firmware combinations as part of their standard testing/deployment process.

bkeith-TT, 2025-02-28:
(Please note: Scripts are still being debugged, and may need specific environment setup to work properly.)

kkabilarTT, 2025-03-10:
Hi, I have the updated scripts with Readme attached to this comment

[tt-umd-scripts.zip](https://github.com/user-attachments/files/19171011/tt-umd-scripts.zip)

bkeith-TT, 2025-07-21:
@kkabilarTT / @tt-rkim Is this done/still needed/superseded by the docker work?

closed 2025-07-21 by kkabilarTT