tenstorrent/tt-umd #571: Refactor tests

broskoTT, 2025-03-04:
Consolidate tests. Also leave a README which explains our structure of tests.

api - these should not perform any unit testing. These are example codes.
gs/wh/bh - arch specfic tests
unit - or some other name - unified, arch agnostic tests

broskoTT, 2025-03-26:
In addition:
- api/unit tests should not be arch specific, and should not have l1_address_map.h available.
- Figure out test namings, make them uniform
- Extract constants and commonly used code like writing our popular 1,2 ... 10 vector to a common test infra file.

broskoTT, 2025-04-08:
Also think about a folder with "benchmark" tests
There is something on brosko/lock_test branch

pjanevskiTT, 2025-04-12:
I think we should also have a group of tests to verify low-level device stuff. For example DRAM speed, eth speed, AICLK...all the values of the subsystems that we believe to be true. Basically the test group should say, ok the card is usable and stable. There was a requirement from syseng folks to provide groups of tests similar to this

broskoTT, 2025-06-30:
Update https://github.com/tenstorrent/tt-metal/issues/24359 if needed once this is done