tenstorrent/tt-umd #585: Blackhole machine broken

broskoTT, 2025-03-06:
Looks like we don't reset it when starting the machine, I synced offline with Vladimir.
When he tried to run tt-metal-smi, the machine completely failed.

broskoTT, 2025-03-25:
Affects whole tt_metal.
Tracked by
https://github.com/tenstorrent/cloud/issues/4185
https://github.com/tenstorrent/tt-metal/issues/19579

vmilicevicTT, 2025-04-03:
Discussed with @broskoTT - Impacts development, as we don't have a functioning BH machine in CI.  Any BH changes need to go through manual testing, increasing the risk of breaking existing functionality.

broskoTT, 2025-04-03:
Copying my comment from the cloud issue:

We had a perfectly happy machine that started failing at one point without any of our own code changes, and has been failing consistently since. It seems like the host machine is crashing rather than just the card hanging:

`The self-hosted runner: umd-p150-1 lost communication with the server. Verify the machine is running and has a healthy network connection. Anything in your workflow that terminates the runner process, starves it for CPU/Memory, or blocks its network access can cause this error.`

The machine is tt-metal-ci-vm-179
Most recent pass observed: 2025-03-05T15:53:21.7310684Z
First failure observed: 2025-03-06T09:43:34.3710146Z

broskoTT, 2025-04-09:
For tracking purposes, I opened a very explicit request here: https://github.com/tenstorrent/cloud/issues/4336

smehtaTT, 2025-04-10:
FYI @broskoTT @vmilicevicTT - Teijo and Cloud team have this BM machine ready - https://github.com/tenstorrent/cloud/issues/4317

closed 2025-04-11 by broskoTT