tenstorrent/tt-umd #600: Deterministic logical ID

broskoTT, 2025-03-11:
It was reported that logical IDs can change from run to run. This was specific for 6U wormhole galaxy, but it might be a deeper problem. 
Investigate code and change if needed so that our IDs always map the same to the chips on a system which doesn't change its configuration.

broskoTT, 2025-03-11:
Allan Liu
  Today at 3:29 PM
[@Bojan Roško](https://tenstorrent.slack.com/team/U06C36SEJ2V)
 the chip ids are deterministic from UMD side, it's the PCI enumeration that happens when you do a reset of the UBB that is the problem.

Bojan Roško
  Today at 3:37 PM
Aha, got it. Would you then consider an improvement if the logical chip ids are deterministic even if the pci device enumeration is random? maybe sort them by some board/asic id?

Pavle Janevski
  [1 hour ago](https://tenstorrent.slack.com/archives/C07L16Z59AP/p1741704134839929?thread_ts=1741641337.938589&cid=C07L16Z59AP)
we have something similar captured in the issue. We can open a separate one if it requires more thinking/discussion. https://github.com/tenstorrent/tt-umd/issues/137#issuecomment-2706549717

Pavle Janevski
  1 hour ago
advantage of 6u over old wh boards is that we have 64-bit unique id that we can read from the chip now
New

Allan Liu
  29 minutes ago
[@Bojan Roško](https://tenstorrent.slack.com/team/U06C36SEJ2V)
 I think what you suggest would be nice to have. UMD physical ids map to some board/asic id, regardless of the pci enumeration

broskoTT, 2025-06-27:
Can be implemented after #871

pjanevskiTT, 2025-07-22:
closed by #1112