tenstorrent/tt-umd #646: TLBAllocator

broskoTT, 2025-03-27:
As depicted by https://docs.google.com/drawings/d/1-m1azdsBqMA0A6ATYRMfkhyeuOJuGCEI62N5a96LXj0/edit

There should be a TLBAllocator class, which has the same interface as defined by #463 
This one lives inside TTDevice.

As a helper class, there should be a StaticTLBManager, which is there not to bog down the Chip class. This one TTDevice doesn't know about.
The current TLBManager is a mix of two, and should be split.

This tasks contributes to overall completion of #248

broskoTT, 2025-04-02:
TLBManager was moved to chip.
But I won't work currently on TLBAllocator, that should be part of TTDevice and another task

pjanevskiTT, 2025-06-06:
for now PCIDevice allocates TLB handles, we are probably not going to implement an allocator, maybe rename the TLBManager at some point

closed 2025-06-06 by pjanevskiTT