tenstorrent/tt-umd #650: AICLK wait to settle

broskoTT, 2025-03-28:
Just copying the whole thread here

Sofija Jovic
  Today at 2:13 PM
I think there is something wrong with how we set freq for BH after the UMD update, I see 800Mhz being printed out in the test locally and on CI.
Issue: https://github.com/tenstorrent/tt-metal/issues/19833

Sofija Jovic
  [Today at 2:13 PM](https://tenstorrent.slack.com/archives/C07QWT4R69M/p1743167586208339)
@Bojan Roško

Pavle Janevski
  Today at 2:40 PM
I can say what I suspect is happening here. After sending the command to raise aiclk device needs some time to actually do it and update the telemetry from where this information is read. If metal runtime goes and reads it immediately, there is a possibility to read the old value, because the telemetry is still not updated. 
@Almeet Bhullar
 already hit this issue few weeks ago. I think metal runtime can implement reading the aiclk value here with timeout until it's sure that aiclk is raised to the proper value. Another solution is to implement this in UMD, if metal is already reading the aiclk, UMD can gurantee raising the clk with readouts so metal doesn't have to do it

Almeet Bhullar
  Today at 2:43 PM
Ah okay yeah I assumed that by the time we read the clock the value would have settled. I’m okay with lowering the print but Metal profiler and other tools might need to capture the aiclk value as well so we would still need access to read it out

Pavle Janevski
  Today at 2:45 PM
you have the access through get_clocks that metal is using. I am saying we can implement readouts immediately after sending arc msg, so metal doesn't have to worry about it in main Cluster workload path

Almeet Bhullar
  [Today at 2:45 PM](https://tenstorrent.slack.com/archives/C07QWT4R69M/p1743169525401919?thread_ts=1743167586.208339&cid=C07QWT4R69M)
Yes that makes sense to me
:+1:
1

Bojan Roško
  Today at 3:19 PM
I'd say that if metal as our client needs an API which is raise_ai_clk_value_and_wait_for_it_to_settle, we should offer this in UMD.
Also, if metal as our client expects that this is settled once the "start_device" call is complete, we should implement it as such in UMD

Almeet Bhullar
  Today at 4:22 PM
Under normal modes of operation I think Metal would always want ai clk to be raised and settled on startup. During debug/bring up is when we’d want to potentially keep aiclk at some min value

abhullar-tt, 2025-03-31:
For debug sometimes we use tools (e.g. `bh-tool` provided by Ivan Hamer) to set a specific AICLK. Currently would sending arc the "go busy" signal override this? (e.g. user sets aiclk to 800M, opens chip, go busy increases to 1350M or would go busy be capped at 800M)

If we don't cap it at user setting then I think it makes sense to expose `raise_ai_clk_value_and_wait_for_it_to_settle` and Metal can avoid calling this in some debug aiclk mode, otherwise I would prefer that the aiclk value has settled once `start_device` is called

pjanevskiTT, 2025-03-31:
I think we should always settle when changing clock in any way, either through start device or some other API. I like exposing separate API for raising the clock, and then start device can either keep the functionality of go busy, or you can call separate API if needed. But either way changing the clock should guarantee settling on the value, that makes sense. Since if the clock hasn't settled there is a problem we need to debug probably

pjanevskiTT, 2025-04-03:
closed by #663

closed 2025-04-03 by pjanevskiTT