tenstorrent/tt-umd #695: ARC heartbeat check seems to be timing out on Metal

tt-rkim, 2025-04-04:
@asrinivasanTT noticed an error like

https://github.com/tenstorrent/tt-metal/actions/runs/14262197831/job/39976649437

where ARC doesn't seem to respond in time.

```
Timed out after waiting 1 seconds for ARC to respond
```

It's odd that ARC hangs would happen like this as AI SW like Metal shouldn't be doing anything to disturb ARC, really. We're wondering if there was a recent change to the ARC heartbeat check logic or timeout or something like that in UMD, as we didn't really see this before on the cloud VMs or in the Metal CI.

It's possible that because of the device layouts on Cloud and how large these machines are, it's possible we're just going over some threshold (1 second?) for waiting for a response. Note that these large T7000 servers which house the CI VMs have 8 cards, so who knows which card each individual VM is on.

It's non-deterministic from the current info we know.

joelsmithTT, 2025-04-04:
https://github.com/tenstorrent/tt-umd/blob/b2330028b0254e5ee2fedccc7f56efe4732b77f4/device/wormhole/wormhole_arc_messenger.cpp#L57 should say milliseconds
https://github.com/tenstorrent/tt-umd/blob/b2330028b0254e5ee2fedccc7f56efe4732b77f4/device/api/umd/device/cluster.h#L398 should say `timeout_ms = 1000`

joelsmithTT, 2025-04-07:
Fixed in https://github.com/tenstorrent/tt-umd/pull/698

pjanevskiTT, 2025-04-07:
closed by #698

closed 2025-04-07 by pjanevskiTT