tenstorrent/tt-umd #697: Certain unit tests can hang the NOC

joelsmithTT, 2025-04-07:
ApiTLBManager.ManualTLBConfiguration is wrong because the tt_SocDescriptor is constructed without knowledge of the hardware.  Therefore the 0th Tensix core's coordinates are not guaranteed to be valid (i.e. the coordinates may correspond to a harvested core)
```
    tt_SocDescriptor soc_desc = tt_SocDescriptor(
        tt_SocDescriptor::get_soc_descriptor_path(tt_device->get_arch()), tt_device->get_arch() != tt::ARCH::GRAYSKULL);

// <snip>

    for (CoreCoord core : soc_desc.get_cores(CoreType::TENSIX)) {
        tlb_manager->configure_tlb(core, core, get_static_tlb_index(core), c_zero_address, tlb_data::Relaxed);
    }

    // So now that we have configured TLBs we can use it to interface with the TTDevice.
    auto any_worker_core = soc_desc.get_cores(CoreType::TENSIX)[0];
    tlb_configuration tlb_description = tlb_manager->get_tlb_configuration(any_worker_core);

    // TODO: Maybe accept tlb_index only?
    uint64_t address_l1_to_write = 0;
    std::vector<uint8_t> buffer_to_write = {0x01, 0x02, 0x03, 0x04};
    tt_device->write_block(
        tlb_description.tlb_offset + address_l1_to_write, buffer_to_write.size(), buffer_to_write.data());
```

ApiTTDeviceTest.BasicTTDeviceIO is wrong because for multiple PCIe devices it will use the 0th Tensix core coordinates for the 0th chip.  The coordinates of this Tensix may not be valid for a subsequent chip (i.e. that Tensix may be harvested).
```
tt_xy_pair get_any_tensix_core(tt::ARCH arch) {
    return std::make_unique<Cluster>()->get_soc_descriptor(0).get_cores(CoreType::TENSIX)[0];
}

// elsewhere...
    for (int pci_device_id : pci_device_ids) {
        std::unique_ptr<TTDevice> tt_device = TTDevice::create(pci_device_id);

        tt_xy_pair tensix_core = get_any_tensix_core(tt_device->get_arch());

        tt_device->write_to_device(data_write.data(), tensix_core, address, data_write.size() * sizeof(uint32_t));
```

pjanevskiTT, 2025-04-07:
Just take the proper non-harvested core for each chip. The fix is easy, I'm on it

pjanevskiTT, 2025-04-07:
https://github.com/tenstorrent/tt-umd/pull/704

broskoTT, 2025-04-08:
This is now confirmed to be fixed on shared pool

closed 2025-04-08 by broskoTT