tenstorrent/tt-umd #736: Not removed semaphores hanging in the system

dbabokin, 2025-04-14:
After running a number of of tests, I observe not removed semaphores on the system:

```
> ls -al /dev/shm/
total 124
<...>
-rw-rw-r--  1 dbabokin dbabokin   48 Apr 13 23:29 TT_UMD_LOCK.CREATE_ETH_MAP
-rw-rw-r--  1 dbabokin dbabokin   48 Apr 13 23:29 TT_UMD_LOCK.LARGE_READ_TLB_0
-rw-rw-r--  1 dbabokin dbabokin   48 Apr 13 23:29 TT_UMD_LOCK.LARGE_READ_TLB_1
-rw-rw-r--  1 dbabokin dbabokin   48 Apr 13 23:29 TT_UMD_LOCK.LARGE_READ_TLB_2
-rw-rw-r--  1 dbabokin dbabokin   48 Apr 13 23:29 TT_UMD_LOCK.LARGE_READ_TLB_3
-rw-rw-r--  1 dbabokin dbabokin   48 Apr 13 23:29 TT_UMD_LOCK.LARGE_WRITE_TLB_0
-rw-rw-r--  1 dbabokin dbabokin   48 Apr 13 23:29 TT_UMD_LOCK.LARGE_WRITE_TLB_1
-rw-rw-r--  1 dbabokin dbabokin   48 Apr 13 23:29 TT_UMD_LOCK.LARGE_WRITE_TLB_2
-rw-rw-r--  1 dbabokin dbabokin   48 Apr 13 23:29 TT_UMD_LOCK.LARGE_WRITE_TLB_3
-rw-rw-r--  1 dbabokin dbabokin   48 Apr 13 23:29 TT_UMD_LOCK.REG_TLB_0
-rw-rw-r--  1 dbabokin dbabokin   48 Apr 13 23:29 TT_UMD_LOCK.REG_TLB_1
-rw-rw-r--  1 dbabokin dbabokin   48 Apr 13 23:29 TT_UMD_LOCK.REG_TLB_2
-rw-rw-r--  1 dbabokin dbabokin   48 Apr 13 23:29 TT_UMD_LOCK.REG_TLB_3
-rw-rw-r--  1 dbabokin dbabokin   48 Apr 13 23:29 TT_UMD_LOCK.SMALL_READ_WRITE_TLB_0
-rw-rw-r--  1 dbabokin dbabokin   48 Apr 13 23:29 TT_UMD_LOCK.SMALL_READ_WRITE_TLB_1
-rw-rw-r--  1 dbabokin dbabokin   48 Apr 13 23:29 TT_UMD_LOCK.SMALL_READ_WRITE_TLB_2
-rw-rw-r--  1 dbabokin dbabokin   48 Apr 13 23:29 TT_UMD_LOCK.SMALL_READ_WRITE_TLB_3
-rw-rw-r--  1 dbabokin dbabokin   48 Apr 13 23:29 TT_UMD_LOCK.TT_ARC_MSG_0
-rw-rw-r--  1 dbabokin dbabokin   48 Apr 13 23:29 TT_UMD_LOCK.TT_ARC_MSG_1
-rw-rw-r--  1 dbabokin dbabokin   48 Apr 13 23:29 TT_UMD_LOCK.TT_ARC_MSG_2
-rw-rw-r--  1 dbabokin dbabokin   48 Apr 13 23:29 TT_UMD_LOCK.TT_ARC_MSG_3
-rw-rw-r--  1 dbabokin dbabokin   48 Apr 13 23:29 TT_UMD_LOCK.TT_MEM_BARRIER_0
-rw-rw-r--  1 dbabokin dbabokin   48 Apr 13 23:29 TT_UMD_LOCK.TT_MEM_BARRIER_1
-rw-rw-r--  1 dbabokin dbabokin   48 Apr 13 23:29 TT_UMD_LOCK.TT_MEM_BARRIER_2
-rw-rw-r--  1 dbabokin dbabokin   48 Apr 13 23:29 TT_UMD_LOCK.TT_MEM_BARRIER_3
-rw-rw-r--  1 dbabokin dbabokin   48 Apr 13 23:29 TT_UMD_LOCK.TT_NON_MMIO_0
-rw-rw-r--  1 dbabokin dbabokin   48 Apr 13 23:29 TT_UMD_LOCK.TT_NON_MMIO_1
-rw-rw-r--  1 dbabokin dbabokin   48 Apr 13 23:29 TT_UMD_LOCK.TT_NON_MMIO_2
-rw-rw-r--  1 dbabokin dbabokin   48 Apr 13 23:29 TT_UMD_LOCK.TT_NON_MMIO_3
```

It doesn't look like expected behavior. Is it? Nevertheless I'm rerunning tests without issues, so  it doesn't seem critical.

I'm using the latest `tt-metal` (https://github.com/tenstorrent/tt-metal/commit/4f706555c6cfcb35a6a1f69783333c04e20ce683), which uses `tt-umd`(https://github.com/tenstorrent/tt-umd/tree/c7526c96f4c1e4dbb50c93f7d1fe61c443839444) with the latest changes to the locking mechanism.

Please let me know if you need more logs or info to reproduce on your side.

broskoTT, 2025-04-14:
This is expected behavior.
Do you have any problems with these files? Would you prefer they're in a folder, e.g. /dev/shm/umd/ ?

dbabokin, 2025-04-14:
Well, my point is that it's unusual to have semaphores left in `/dev/shm/` after the application execution completed. The application is expected to clean up after itself after it's done. So why are the semaphores are not closed on the exit?

Basically, whenever I have problems with TT device - any crashes, not 100% reproducible behavior, etc. I start to suspect that something in the environment caused that, and these semaphores is suspect # 1 for me.

Also looking at these semaphores I see that they have `rw` for the user and the group, but `r` for all. What supposed to happen when the other user, who is not in the same group will try to use the device after I completed the execution?

dbabokin, 2025-04-14:
Ok, this sounds reasonable. Feel free to close that.

dbabokin, 2025-04-15:
Well, these semaphores without write permissions for all become a problem for us. We have multiple users on the machine, including CI. So if the user haven't cleaned up after themselves, then another user has a problem. And even with sudo permissions it's not save to remove the semaphores, as the device might be really in use.

One other issue is that when running in the docker, `/dev/shm` is not shared with the host by default.

dbabokin, 2025-04-15:
Our current solution is to remove these semaphores on our side in the context destructor - i.e. when the job is completed. But it's really a workaround, it's not ideal.

broskoTT, 2025-04-15:
> these semaphores without write permissions for all become a problem for us

The referenced PR should fix this problem

> One other issue is that when running in the docker, /dev/shm is not shared with the host by default.

Do you have a scenario of multiple users using multiple dockers running on the same set of chips? We don't support that for now. If not, please describe the scenario on why these semaphores not being shared between dockers is a problem

dbabokin, 2025-04-15:
> Do you have a scenario of multiple users using multiple dockers running on the same set of chips? We don't support that for now. If not, please describe the scenario on why these semaphores not being shared between dockers is a problem

I don't fully understand the role of these semaphores, so I don't know if this is going to be a problem or not. But two users running their tasks in the containers will not be able to detect the other user, right? What would happen in this case if they try to use the device concurrently?

Alternative solution might be passing through `/dev/shm` to the container. But it will have it's own problems - as the users / user ids might be different on the host and in the container.

broskoTT, 2025-04-15:
> What would happen in this case if they try to use the device concurrently?

I don't think we support this scenario currently. There are many other changes throughout the stack besides these mutexes.

What is supported though is one user (one containers) running multiple processes/threads over the same device.
Also multiple users/containers using multiple devices on the same host system.

dbabokin, 2025-04-15:
Ok, got it, thanks.

broskoTT, 2025-04-15:
@dbabokin thanks for surfacing this issue and thanks for the feedback! I'll close this issue, I merged the referenced PR (this is not yet in tt_metal though). If there is something else to discuss further, feel free to reopen the issue.

closed 2025-04-15 by broskoTT